CaptainBobSim/The-Cessna-172-Project-V3 #134: Location of Indicators?

Kovinator, 2022-08-20:
Indicators or instruments like turn coordinator are in section 2A while airspeed indicator and generic are in section 2. Section 2 is titled an instrument section while section 2 says the analog panel.  I don't know if this is intentional or not but I would think it would make more sense to have all the analog dial indicators aka steam gauges  (both small and large) be in one folder and the rest like throttle flaps etc in the analog panel folder.  Your thoughts? Btw I have another pull request coming soon - my focus has been on the support for the  LED lighting to all instruments as well as the 28BYJ stepper motor but would like some clarification as to where these should be located before I contribute. Thanks!

CaptainBobSim, 2022-08-21:
I have these in separate folders because Section 2 is all of the shared components (Airspeed Indicator, Structure and Flaps are the same parts between the G1000 model and the analog model). Right now I only have an analog model so this doesn't make sense, but I have G1000 plans in the future.

What are your thoughts with this?
Is there a better way to do it?

I think your idea of having all steam gauges in one folder is awesome. Where do you think this should go?

Kovinator, 2022-08-21:
Ok, that makes sense. Thanks for the quick reply and clarification! My thoughts on this:  

**Option 1:** Leave the current structure as is and simply update the "What Does Each Section Mean?" under the code section.

**Option 2:** Leave the current structure as is and just rename a couple of folders. Ex:
  
Section 2 - Glareshield and Shared Instruments (G1000 and Analog)
Section 2A - Analog Panel and Instruments
Section 2G - G1000 Panel and Instruments

**Option 3:** Change the directory structure to combine and simplify: 

Section 2 Glareshield and Panels 
Section 3 Instruments
Section 4 Controls
Section 5 Electronics

With this structure, everything is combined and clearly defined. When going into each section people should be able to figure out what goes on what. 

For example, in section 3 the turn coordinator or heading indicator is only on the analog panel.  Note there are only three instruments that are actually shared between the G1000 and analog: The Altimeter, Vertical Speed, and Attitude indicators. Everything else is on the Garmin. 

Section 4 would have the physical hands-on controls  (flaps, throttle, yoke, switch panels, etc).  

Section 5 electronics. Again the idea here is people can figure out what goes where (Radio Stack equipment on analog, G1000 on the Garmin Panel, annunciator panel on analog, etc. You get the idea.

It's your call how you want to proceed and now that I think about it if you decide to make changes I should commit my changes first while our directory structures are the same that way there are no conflicts. Let me know thanks again!

CaptainBobSim, 2022-08-23:
I decided to go with a mix of all three

I mostly adopted option 3, but with different names.

Section 2 refers to the space on the floor being taken up, so I named the folders 2-(number)-XXX for the level of organization.

[image]

If you have any improved names for any folders, don't be afraid to shout them out!

Kovinator, 2022-08-25:
I like the great rename! I didn't mean to create more work for you. I was happy with the clarification you gave but man, with what you've done it's much cleaner now! I really like how you did your own combination by adding an extra level of abstraction with section 2 being simply called components and with everything under that being broken down into the appropriate sub-components. Nicely done!

closed 2022-08-25 by Kovinator

CaptainBobSim, 2022-08-26:
Not extra work! Less work! Better to do it now than when everything is populated.
Thank you for sharing your ideas! I like this system much better :)