CaptainBobSim/The-Cessna-172-Project-V3 #155: EGT Indicator Question

Kovinator, 2022-10-10:
How does the EGT knob feel when you turn it in the real aircraft? Does it click as you rotate the knob thus moving the needle or is it smooth? Does anyone know? I'm almost done with the 9g servo version of the EGT just need to know this to complete it. If we don't know then I'll probably make it smooth because it would be an easier design. I already have a clickable version but it is going to be tricky to work out so that it clicks and thus steps properly on each line on the decal.

CaptainBobSim, 2022-10-10:
my assumption is that it is smooth, but I will check my PPL course to be sure.

CaptainBobSim, 2022-10-10:
It looks like it can be between lines: 
[image]
It shouldn't have to snap to lines

closed 2022-12-16 by CaptainBobSim