CaptainBobSim/The-Cessna-172-Project-V3 #196: Figure out "Open source" vs "Proprietary"

CaptainBobSim, 2023-01-14:
If I develop something I prefer to have it accessible for free but not open source. I plan to eventually open-source everything but would like to first be in a financially stable spot.

If someone designs something it is  "Open Source" - anyone can use something

If you don't want me (or anyone else) to be able to sell it use the "DO NOT SELL" 

You may sell things you develop.

Kovinator, 2023-01-15:
My two cents - You may want to look into a dual license or a copyleft license (such as the GNU GPL) which can provide limitations so that people can't sell or use it for commercial use but can still copy distribute modify etc. That way you could still provide the source but have a license that prohibits anyone from selling it. Most people don't consider this open source but it's in between the open source and proprietary. Just a thought.

CaptainBobSim, 2023-01-15:
Currently, it's licensed under "Creative Commons Attribution-NonCommercial 4.0 International Public License"
Which should allow modification and distribution, but prevent commercial use (other than myself).

For contributors do you think it is fair to sell things that I don't design myself?

What do you think of this system: 
https://github.com/CaptainBobSim/The-Cessna-172-Project-V3/tree/main/Section%200%20-%20Info%20and%20Full%20Assembly/FOLDER%20LEVEL%20LICENSES%20-%20for%20contributers

A contributor can add their file to change the license for their folder only.

I'd have to make it all "up to snuff" https://opensource.stackexchange.com/questions/2692/how-to-denote-certain-files-are-under-a-different-license

Kovinator, 2023-01-15:
Ah so you already have a license, I never checked and assumed it was open-source. This is good to know. So what you are really after is on the contribution side of things. I guess there really is no right or wrong way to do this as long as it's easily understood. To answer your question: "For contributors do you think it is fair to sell things that I don't design myself?" my answer is yes but it's nice if you ask for permission.  Giving the contributor the option is exactly that - asking for permission but it goes a step further allowing them to make it not for sale (you included) or make their work truly open source for everyone. Your proposed system I believe does this in an easy-to-understand manner and I commend you for trying to make this as flexible as possible for everyone. Well done!

closed 2023-01-16 by CaptainBobSim