CaptainBobSim/The-Cessna-172-Project-V3 #197: Junction Disfunction is not a function!

Kovinator, 2023-01-15:
I don't know if this is reproducible or something weird I don't understand but the junction boxes I made to consolidate common wires is causing some problems. The issue is that at least for servos I'm seeing what can only be described as feedback jitters. 

For example from the sim I can fill the tanks, when I fill the right tank in the sim the left tank goes up and down like crazy really fast even though that tank isn't being modified. If I look at mobiflight I do not see the numbers change for the left just the right so it's not the sim or mobiflight.  If I take the 5v and ground for that servo and plug it into a breadboard instead of the junction the problem goes away. Have you seen this before? Do you know what is causing it? I wonder if there is RF noise being generated due to the servo wires being really short (only an inch or two). Or maybe it has something to do with junction boxes being too close to the servo?  Maybe it's both? Maybe the current draw is too high and its causing signal drops? (don't have external power supply yet) and I'm driving 4 servos with the Arduino. Maybe I'm somehow generating an inductive field that will open a gateway to another dimension? Maybe aliens are trying to speak to me through the servos somehow.  Got any thoughts?

CaptainBobSim, 2023-01-15:
I think the Arduino can provide 500mA via USB, and each servo uses somewhere around 250mA. 

Try running the test sketch on motif light which will trigger them each individually vs trying to move them all at once.
Also, consider trying to run just two at a time in your current configuration and see if that changes anything.

I'd also prepare a light snack of various electronics for the aliens - a long voyage might make them power hungry!

Kovinator, 2023-01-15:
Yeah already tried just the two servos. I’ll try your suggestions thanks !

Kovinator, 2023-01-20:
Update. It seems to be related to the LEDs. Specifically PWM. If the LEDs are on using PWM then the servos seem to go crazy. For example:

LED Transform:
if (1=1,255,0) LEDs ON - NO PROBLEM
if (1=1,0,0) LEDs OFF - NO PROBLEM
if (1=1,128,0) LEDs (50%) Crazy servo moving all over the place not obeying mobiflight.  

Pinout:
led pins (5) PWM enabled in mobiflight Note: All LEDs share this pin
LEDs are wired in parallel each having its own 100ohm resistor wired to the cathode. 
servo pins (31,33,35,37) are signals for the servos.

Power:
I'm now using an external power source (modded PC power supply). This will go inside the glovebox to power the entire panel. All grounds and voltages are supplied by this power supply to everything using a breadboard.
The Arduino is powered from this power supply through the VIN pin and ground using the same power supply. 

Conclusion:
I must have miswired something. I suspect one of the LEDs is to blame. I could have wired it backward? The resistor on positive instead of negative maybe? One of my servos quit working which I'm not sure if is related or not.  I'll keep at it.

Kovinator, 2023-02-21:
There are actually two issues and both now are solved. The servo jitter originally was caused by pulling too much amperage when the Arduino is powered just by USB as Trevor suggested. When using an external power supply to try and fix this the jitters became worse and I assumed this was the same issue but this was actually caused by a separate issue related to incorrect wiring of the external power supply.  

Using an external power supply is a must when having a higher load. When using an external power supply I had positive 12v and negative hooked to the power pins on the Arduino which is pretty much the same as plugging it in through the DC jack. This powers the Arduino and allows for more amperage (1 AMP as opposed to 1/2 AMP USB).  Also, it allows me to power servos and step motors directly from the power supply to keep the load on the Arduino as small as possible.

Resolution:
My mistake was grounding the breadboard to the common ground on the power supply. Seemed logical to me at the time but it's not. It allows for different ground potentials. To correct the issue the breadboard needs to be grounded to the Arduino NOT the power supply.

Simple Servo example -  (Breadboard eliminated for simplicity) What I had was this:
Servo Positive - Power Supply +
Servo Negative - Power Supply -
Servo Signal - Arduino Servo Pin (PWM)

I changed it to this:
Servo Positive - Power Supply +
Servo Negative - GND pin on Arduino
Servo Signal - Arduino Servo Pin (PWM)

A simple mistake that took hours to find. Hope this helps someone else. Eventually, I would love to see a step motor design for the smaller instruments but they are too small to put the step motors directly side by side. Eventually, we will get there! I'm glad this is finally closed. Thanks for the help!

closed 2023-02-21 by Kovinator

CaptainBobSim, 2023-02-23:
I'm glad you could solve this! I will make sure to be careful when wiring ext power supplies.
Thank you for sharing your findings!