CaptainBobSim/The-Cessna-172-Project-V3 #307: Altimeter Wheel Gear Retainer

feldon30, 2023-06-01:
A possibly modification to the Altimeter assembly would be to add two indentations to the Faceplate on the center column and then create a retaining ring that's 27mm OD, 11mm ID, 2mm thick with 2 pegs that would be pushed into (or glued) to the faceplate. The Wheel Gear would be able to spin freely but would be held captive to the faceplate.

CaptainBobSim, 2023-06-06:
I put in a retainer attached to the 10K gear, we'll see how this works.
[image]

I didn't want to glue it because I'm trying to avoid glue in this project to be maintenance friendly.
I couldn't screw the retainer in without using M2 screws which are also being avoided because they are generally annoying to work with.

CaptainBobSim, 2023-06-07:
Tested and works well!
Does not interfere with card gear.

feldon30, 2023-06-07:
Awesome!

CaptainBobSim, 2023-06-07:
Yep!