CaptainBobSim/The-Cessna-172-Project-V3 #316: Upgrade Six Pack Instrument Graphics

feldon30, 2023-07-02:
Six Pack instrument graphics seem to be derived from different origins in different qualities.
Recreate/trace graphics into scaleable vector format.

CaptainBobSim, 2023-07-02:
Hello! All of the 6-pack instruments except the Airspeed are vectorized. Any quality issues are from export.
Would you want to trace/replicate the Airspeed Indicator in an editable SVG format?

Graphic design is done in Inscape (open source tool) to make collaboration not locked into a paid platform. If needed you can do it in Adobe Illustrator.

The template for Inkscape SVGs is in 
The-Cessna-172-Project-V3\Section 0 - Info and Full Assembly\--- COMPONENT TEMPLATE\___Decals.svg

CaptainBobSim, 2024-10-05:
I believe most are "die cut" format now, feel free to speak up if I missed any.
Thank you again @feldon30 for making these AMAZING decals!