CaptainBobSim/The-Cessna-172-Project-V3 #446: Pin 13 Strange behavior

CaptainBobSim, 2024-03-28:


oOblik, 2024-04-06:
This is a known thing https://docs.arduino.cc/learn/microcontrollers/digital-pins/

> Digital pin 13 is harder to use as a digital input than the other digital pins because it has an LED and resistor attached to it that's soldered to the board on most boards. If you enable its internal 20k pull-up resistor, it will hang at around 1.7V instead of the expected 5V because the onboard LED and series resistor pull the voltage level down, meaning it always returns LOW. If you must use pin 13 as a digital input, set its pinMode() to INPUT and use an external pull down resistor.

CaptainBobSim, 2024-06-06:
This has been fixed with the redesigned Breakout Board System (V2.0)

closed 2024-06-06 by CaptainBobSim