CaptainBobSim/The-Cessna-172-Project-V3 #448: Radio: Did you notice you have NAV labeled as COMM in your designs?

CaptainBobSim, 2024-04-05:


oOblik, 2024-04-05:
To add to this, as I put the radio together there are a few other minor things I found that you may or may not want to update (they are very nitpick-y I know):

Having a notch to accommodate the little pin that sticks out of the front of the dual-encoders that prevent them from rotating when used would be nice. Right now they press against the back of the front plate and kind of force them to sit crooked. It also makes it much harder to properly tighten them down.

On the PropWashSim Dual 5 Digit display, the header isn't completely behind the segment LEDs and the solder points on the other side of the PCB prevent it from sitting flush on the enclosure. Having a small (~1mm) indent in between the displays to accommodate the solder points would solve that.

CaptainBobSim, 2024-04-06:
Hey @oOblik Could you draw these out, I'm having a hard time understanding exactly what you're saying

CaptainBobSim, 2024-04-08:
Ahh, thanks for showing me! Usually those would be clipped with side snips but it would be a good design feature to put in a cavity (especially for the encoder - to stop bad rotation)