CaptainBobSim/The-Cessna-172-Project-V3 #498: Make Buttons fo 530 kit without labels

ecdcrew, 2024-09-20:
For kit buyers, 80% (or more) will benefit from labels (OBS, VNAV, FPL etc) embossed on the button itself, as you have done. However, for those of us with Lasers (CNC) it would be cool to have no raised labels on buttons. Please consider this a VERY low priority item (but you can apply the logic to all buttons in all kits).

I print in white, spray paint black, then let the laser do the magic or removing the .3mm of black to reveal detailed text.

CaptainBobSim, 2024-09-20:
If you ever want to make an inkscape file with the labeling I'd be down to include this in the project files! 
Regardless I'll put it in my someday list :)

ecdcrew, 2024-09-20:
Actually, I meant the STL button, not a label file. Literally, your tall and short buttons have text on them for the 530 kit, I'm saying leave them as plain buttons.

RE: [Button_tall.stl](https://github.com/CaptainBobSim/The-Cessna-172-Project-V3/blob/main/Section%202%20-%20Component%20Library%20and%20Structure/2-6%20Avionics%2C%20Radios%2C%20Annunciators/GPS_GNS%20530/STLs/Buttons_Tall.stl) etc.

CaptainBobSim, 2024-09-20:
For sure! I've just exported the STLs (I just performed a cut in Bambu Slicer), if you have an SVG file to send to a cutter I think that could be helpful for people? If that's something you want to make it could be super dope! but if not I can do it sometime :)

CaptainBobSim, 2024-09-20:
Only if you'd already do it or if you want to though!